LorinczElectronics/Orbiter-Toolboards #4: No mounting option for PCB board O2 toolboard kit

Johan501, 2024-07-13:
It seems that there isnt any way to mount the USB PCB somewhere?

Johan501, 2024-07-15:
I litterally can't continue my LDO Voron 2.4 rebuild for days now.... so I'm pretty frustrated to be really honest.

RobertLorincz, 2024-07-15:
The only design I've made is to attach it to the back of the stepper motor. The fixing holes are also designed in a way to fit there perfectly. This does not mean you cannot mount it anywhere else. You would just need some adapters or a way to fix it where it is better suited for you.

Johan501, 2024-07-15:
> The only design I've made is to attach it to the back of the stepper motor. The fixing holes are also designed in a way to fit there perfectly. This does not mean you cannot mount it anywhere else. You would just need some adapters or a way to fix it where it is better suited for you.

Hi, thanks for your reply, but are you talking the toolhead board itself?
I was talking about the USB PCB board, not toolhead board. That was has 2 brackets. But I digress.

RobertLorincz, 2024-07-15:
I see, I misunderstood your issue. I thought you are talking about the Toolboard. Now I see you mean the small USB adapter PCB. Give me a min and I upload the design I use on my printer.

Johan501, 2024-07-15:
Hi,

It's okay!
And thank you very much!
I saw on Twitter that someone else also had the same issue and probably
others maybe too.

Take your time, I'm already glad you will upload it! :)

Today I made one myself, not sure yet if it will work. I probably going to
use yours. But in general it's better to have it uploaded. :)

Kind regards,
Johan

RobertLorincz, 2024-07-15:
I uploaded my version together with the adapter board CAD model. I also included the Fusion 360 and .step files, so people can modify as they wish; maybe my mount will not fit everyone. I also updated the toolboard user manual.

Johan501, 2024-07-15:
Okay awesome, thank you! 😊🤘🙏

closed 2024-07-16 by Johan501