LorinczElectronics/Orbiter-Toolboards #8: Problem flashing klipper with katapult bootloader to SO2

jpapiez, 2024-09-06:
I want to install the Katapult bootloader to make it easy to update the klipper firmware, then deploy updates to klipper firmware via katapult. I flashed katapult with the following options:

Enable low-level configuration options
Micro-controller Architecture: STM32
Processor: STM32F042
Build Katapult deployment application (Do not build)
Clock Reference (8MHz crystal)
Communication interface: USB (on PA11/PA12)
Application start offset: 8KiB offset

Then I flashed Klipper with the following options:
Enable low-level configuration options
Micro-controller Architecture: STM32
Processor: STM32F042
Bootloader offset: 8KiB bootloader
Clock Reference (8MHz crystal)
Communication interface: USB (on PA11/PA12)
Optional features:
* Support GPIO "bit-banging" devices
* Support external sensor devices
* Support lis2dw 3-axis accelerometer

brunoR1000, 2024-09-06:
This might help you
https://bit.ly/3yXXzbX
Passcode: d5kp14h
You must have gcc compiler installed

RobertLorincz, 2024-09-08:
You did not mention your issue.
In the past, I tried this as well but there is not enough memory to do that in the STM32F042.

RobertLorincz, 2024-12-01:
Unfortunately, Katapult + Klipper do not fit into this microcontroller version; I'm working on a new toolboard version and using a bigger micro there.

closed 2024-12-01 by RobertLorincz