LorinczElectronics/Orbiter-Toolboards #10: Lost communication with MCU 'toolboard_t0'

dyocis, 2024-10-07:
I'm running RatRig V-Core4 500 with a Orbitool board.  I keep getting  lost communication error.   I've tried multiple toolboards, USB pcbs, and cables.  I ordered 3 toolboards, so I had 1 for IDEX and a spare.  

It seems to work fine for 4 or 5 prints, then I get 3ish hours into a print and it shuts down due to "Lost communication with MCU 'toolboard_t0'" Then it does it every time I try to print something until I switch out the board or cable.  Then it will work fine for 4 or 5 prints.  I'm really stumped on this.  All the connectors are stock, I didn't crimp anything.  I've switched out previously removed boards and cables, and it will still print 4 or 5 prints without issue.

T/S info:

RatOS 2.1
Changed stock toolhead cables - I have rotated out 3 different stock cables
Changed USB cables - rotated out 3 stock cables, as well as a known good usb 3.1 cable
Verified 24v  - 24.3v while printing
Reported temp while printing: 70-75c 
The VC4 part cooling fan draws air across board while printing, so there is a constant flow of air across the heatsink.

I can provide the klippy log, but it's too big to attach here.

RatOS.cfg:
[board_pins toolboard_t0]
mcu: toolboard_t0
aliases:
	#------------------------------------------- EXTRUDER motor pins --------------------------------------------
	e_step_pin=PB7,
	e_dir_pin=PB6,
	e_enable_pin=PB4,
	e_uart_pin=PB5,
	#----------------------------------------------- GENERAL PINS -----------------------------------------------
	e_heater_pin=PA0,
	e_sensor_pin=PA3,
	stepper_spi_mosi_pin=null,
	stepper_spi_miso_pin=null,
	stepper_spi_sclk_pin=null,
	adxl345_cs_pin=PA4,
	bltouch_sensor_pin=PA15,
	bltouch_control_pin=PB3,
	probe_pin=PA15,
	fan_part_cooling_pin=PA10,
	fan_toolhead_cooling_pin=PA9,
	fan_controller_board_pin=null,
	heater_bed_heating_pin=null,
	heater_bed_sensor_pin=null,
	4p_fan_part_cooling_pin=null,
	4p_fan_part_cooling_tach_pin=null,
	4p_toolhead_cooling_pin=null,
	4p_toolhead_cooling_tach_pin=null,
	4p_controller_board_pin=null,
	4p_controller_board_tach_pin=null

[mcu toolboard_t0]
serial: /dev/RatOS/ldo-orbitool-o2-t0

[temperature_sensor Orbitool_O2_T0]
sensor_type: temperature_mcu
sensor_mcu: toolboard_t0

[lis2dw toolboard_t0]
axes_map: x, z, y # Assumes back-facing vertical toolboard mounting
cs_pin: toolboard_t0:PA4
spi_bus: spi1

[extruder]
max_power: 0.995

[adc_temperature hotend_power_scale]
temperature1: 119
voltage1: 1.91
temperature2: 239
voltage2: 3.82

[fan]
max_power: 0.995

[heater_fan toolhead_cooling_fan]
# Use max_power: 0.5 for 12V, max_power: 1 for 24V (in printer.cfg)
max_power: 0.995
cycle_time: 0.0001

[output_pin O2_RUN_LED]
pin: toolboard_t0:PB8
shutdown_value: 0
value: 1

[temperature_sensor hotend_power_draw]
sensor_pin: toolboard_t0:PA2
sensor_type: hotend_power_scale

printer.cfg:

[fan]
shutdown_speed: 0.0
hardware_pwm: False
cycle_time: 0.0001
kick_start_time: 0.2
# off_below: 0.0
max_power: 0.995
#pin: toolboard_t0:PB1

[heater_fan toolhead_cooling_fan]
shutdown_speed: 0
tachometer_pin:toolboard_t0:PA8
tachometer_ppr: 2
tachometer_poll_interval: 0.0015

nickidotdk, 2024-10-25:
Hi @dyocis sorry to hijack this thread, but I am in the process of adding this board to my VCore. How did you wire the RatRig endstop to the board? Since there are only 2 pins, I guess the led feature of the endstop will not work when triggered. Is there anything else you had issues with installing this board?

Best regards

RobertLorincz, 2024-10-30:
can you share klipper log file?

winningfaith81, 2025-02-17:
I run into similar issues with disconnection from the toolboard. I have found that you must use the wire harness because as your printer is moving around the port that the cable connects to moves around. As I am in the process of building my only printer, I can't print one currently. I have used several zipties anchoring it to the Orbitor filament sensor and this has seemed to drastically improve my experience. 

However, I have an issue where if I touch my hotend with something metal (let's say to change my nozzle) the board shuts down and gives a connection error. I wonder if this has something to do with having a Beacon probe? It sounds like the metal I'm introducing is shorting out the connection, but I'm not sure where/how.

closed 2025-11-17 by RobertLorincz