LorinczElectronics/Orbiter-Toolboards #11: O2 PT1000 no selector in documentation

pmarkiewicz, 2024-10-25:
I'm looking to buy O2 toolboard, documentation states that it's compatible with PT1000 but I don't see any resistor selector for PT1000

RobertLorincz, 2024-10-29:
the selection is done only by klipper config

pmarkiewicz, 2024-11-01:
so there is no way to change resistor value? it's a pity as this heavily
reduces precision of PT1000.

regards

pmarkiewicz, 2024-11-04:
> The resistor value is already optimized for PT1000. 2200 Ohm works fine
> for both PT1000 and standard NTC.

Good information, can you update documentation please?

pmarkiewicz, 2024-11-20:
> I do not know where you expect the documentation to be updated. Klipper
> config example shows 2200 ohm same in the user manual....

just toolboard documentation that there is 2.2k resistor.

RobertLorincz, 2024-11-27:
It is written in the documentation check it...
[image]

closed 2024-11-27 by RobertLorincz