LorinczElectronics/Orbiter-Toolboards #12: SO3 COmmunication issues.

Naveedy0, 2024-10-26:
[image]

Im sad and mad that I ever bought this. 
Ever since I changed this, every once and a while it will ruin my print. Sometimes at the beginning other times mid print. Worst is after 12 hours printing time, where it suddenly stops communication. 

I'm using a CB1 klipper setup on a Manta board. 
I don't know what to do anymore.

RobertLorincz, 2024-10-30:
could you share with us klipper log file when this happens? Do you have the connection cable secured properly so it cannot disconnect itself?

Naveedy0, 2024-10-30:
I found out what the problem is. The o3 mcu gets crashes when the
temperature reaches 60+ degrees. It can't be cooled passively I had to add
a fan and change the steps from 32 to 16.

ons. 30. okt. 2024 08.34 skrev LorinczElectronics ***@***.***

RobertLorincz, 2024-10-30:
The microstep setting should not influence the stepper driver temperature...16 or 32 should give the same value. Yes, temperature can be an issue; you could reduce stepper current. It also depends on how much air cooling you have in that region. Currently, we are also investigating this issue since the board is designed to work at higher temperature than 60°C. Mines are running fine at over 80 °C. It seems we got some bogus parts we use for the USB communication protection, which does not work well in hot environment as designed. If the issue persist we can try replace your unit.

Naveedy0, 2024-10-30:
I mean I would like a unit that could run passively.
My ambient temperature is 24 degrees, when I bought it I had a
understanding that that would be okay.

ons. 30. okt. 2024 09.04 skrev LorinczElectronics ***@***.***

Naveedy0, 2024-10-31:
> So I changed it back to 32steps. And the so3 lost communication again
> during print.
> This time it wasn't the temp that was the issue. I don't think it can
> maintain 32steps.
> I have attached the log file.
>
> Den ons. 30. okt. 2024 kl. 09.06 skrev Naveed Yousaf ***@***.***

I tried again, 16 steps, no fan and the temperature reached 64 degree and
it stopped SO3.

Naveedy0, 2024-10-31:
And this time, with 16 steps and fan on, and it still stopped working.
I can't use this :(

Naveedy0, 2024-11-02:
I have decided to remove it and run the orbiter without the ldo.
It is completely useless.

RobertLorincz, 2024-11-04:
you may contact your reseller and ask for a replacement. Your unit is probably defect.

thomasgalambos1, 2024-11-26:
I have the same problem..... it is installed on a sv08 and it did run for aprox 10 prints and now every print crashes the same way... The orbiter 3 is great at printing but the combo with the toolboard is useless...

Naveedy0, 2024-11-26:
> I have the same problem..... it is installed on a sv08 and it did run for aprox 10 prints and now every print crashes the same way... The orbiter 3 is great at printing but the combo with the toolboard is useless...

I returned mine and rewired everything. 
The only issue is the mess of a wire other than that everything works now.

RobertLorincz, 2024-11-27:
We had some bad component batches related to the USB protection circuit. We are working to fix it. I can only recommend users who have such issues to check my wiring guides on the orbiterprojects webpage. Ask for a replacement board along with the small adapter board if that does not resolve the problem, as that might also be the cause. 
One big influence can be the supply voltage for the small adapter board. Make sure it does have 5V power supply from the USB connector, not less. It is common that this issue is due to a bad 5V power supply performance; instability due to load current change will cause a communication crash. The new version of the adapter board will completely fix this issue, it will have its own 5V power supply and not rely on the voltage from the RPI which can be dirty.

Naveedy0, 2024-11-27:
> We had some bad component batches related to the USB protection circuit. We are working to fix it. I can only recommend users who have such issues to check my wiring guides on the orbiterprojects [...]

Please tell when the new version gets released. Whats the name going to be of the new version? 
- I would like to have a pretty printer again.

RobertLorincz, 2024-11-27:
We are having the first samples now. I could arrange for you to get a sample. For me at the moment, it is not super clear if the issue is on the toolboard or on the adapter board. We know we had a bad batch on one component in the USB protection circuit. But its not fully clear what reached customers. I have seen many issues where the actual printer setup caused the issue with the 5V supply of the adapter board. This is very tricky for users to spot. Would you be able to help us debug this more deeply? I could arrange for you a new adapter board with its own 5V power supply, and you may check if the issue still persists or not with the toolboard you have.

thomasgalambos1, 2024-11-27:
> We are having the first samples now. I could arrange for you to get a sample. For me at the moment, it is not super clear if the issue is on the toolboard or on the adapter board. We know we had a [...]

I am fine with that because except of this issue orbiter 3 and the toolboard deliver very best prints. I have the toolobard on sv08 which has a mks skpir base dboard but i also checked raspberry pi and have same situation

Naveedy0, 2024-11-27:
> We are having the first samples now. I could arrange for you to get a sample. For me at the moment, it is not super clear if the issue is on the toolboard or on the adapter board. We know we had a [...]

I would like to try it out as well. It would give me a peace of mind as well to know the reason behind why it wasn't working properly.

thomasgalambos1, 2024-11-30:
> I am fine with that because except of this issue orbiter 3 and the toolboard deliver very best prints. I have the toolobard on sv08 which has a mks skpir base dboard but i also checked raspberry pi [...]

On a sv08 i wired the original cable (24v, gnd, d+,d-) from motherboard toolohead connector  which means no usb adapter board bentween orbiter toolboard and motherboard and i am still testing on a longer print BUT it looks like the toolboard mcu is constnant at 54C and no more crash so 90% sure that the usb adapter is the  problem .. but i also adjusted main power source to be exactly 24v.... (was little bit less) . At least on sv08 ... this is the situation

Naveedy0, 2024-11-30:
Will it help if I just added a little fan to cool down the usb adapter? Maybe it just gets to hot inside the kabinet?

RobertLorincz, 2024-12-01:
> On a sv08 i wired the original cable (24v, gnd, d+,d-) from motherboard toolohead connector which means no usb adapter board bentween orbiter toolboard and motherboard and i am still testing on a [...]

Yes, it is very possible you are correct. LDO discovered they have received a fake or bad batch of components that are used in the USB protection circuit. Replacing the adapter board solved the issue I confirmed in a few cases I checked.

thomasgalambos1, 2024-12-01:
> Yes, it is very possible you are correct. LDO discovered they have received a fake or bad batch of components that are used in the USB protection circuit. Replacing the adapter board solved the [...]

Great, how can we  get a new adapter board?  I would also like to purchase several orbiter 3 and toolobards but now i am ... stuck because i don't know how to purchase the fixed version"

RobertLorincz, 2024-12-03:
The fixed adapter version will be out soon; you can request a free replacement from LDO or your reseller. I do my best to assist here.

cyablo, 2025-02-28:
> The fixed adapter version will be out soon; you can request a free replacement from LDO or your reseller. I do my best to assist here.

Does this also apply to the Orbitool o2? I had a hell of a time with prints stopping & RPi freezing, tried Kernel, cables, power supply and stuff and now after the latest freeze, the o2 doesn't even show up as as USB device and seems to be dead!

RobertLorincz, 2025-03-01:
Yes, since both have the same adapter board. if that does not solve the issue, you should request a replacement board.

RobertLorincz, 2025-03-07:
Yes, it applies to Orbiter O2 toolboard as well. In case you are experiencing the issues like you mention it also can be that the toolboard has some issues, please ask for a replacement board from your reseller, LDO will support fixing the issue.

Dr. Robert Lorincz

supercrossed, 2025-03-10:
Hi Robert, just came across this thread and having the same issue - intermittent MCU disconnects with the O2, and now the nitehawk36. The nitehawk is the 'updated' release from earlier this year, and the USB board is different from the O2, it is about twice the size. Is this the updated version?

How could I request an updated 5v adapter board?

closed 2025-03-10 by Naveedy0