LorinczElectronics/Orbiter-Toolboards #15: FIlament runout on SO3 Toolboard not working

Luca432122, 2024-11-09:
[image]

I have not made any changes to the runout config but i keep getting the error "The value 'runout_filter' is not valid for ID"

Luca432122, 2024-11-10:
I've just noticed that runout is written "runout" in the switch definition and "runnout" in filament macros could that be the issue?

`#***********************************filament swicth definition***********************************************************

[filament_switch_sensor SO3Senzor]
switch_pin: SO3:PA13
pause_on_runout: False # do not change to true, printer is paused in the next sections
runout_gcode: UPDATE_DELAYED_GCODE ID=runout_filter DURATION=0.5 # duration defines the filter time for runnout detection
insert_gcode: filament_load_init
event_delay: 1.0
pause_delay: 0.1

#////////////////////////////////////////Filament Unload macros/////////////////////////////////////////////////

[gcode_button filament_unload]
pin: !SO3:PA10
release_gcode:  # filament unload procedure   
  SET_LED LED=SO3RGB RED=0.0 GREEN=0.5 BLUE=0.5
  filament_unload_init
press_gcode: # do not add any macro call here

[delayed_gcode runnout_filter]
gcode: #checking if filament runout is still present after defined delay time, default 0.5s
  {% if printer["filament_switch_sensor SO3Senzor"].filament_detected==false %}
      runnout_init      
  {% else %}
      M118 Runnout sensor intermitent contact detected!
  {% endif %} 
`

Luca432122, 2024-11-10:
Happened due to this #1

RobertLorincz, 2024-12-01:
this bug is corrected in the new macro of the extruder.

RobertLorincz, 2024-12-01:
The new macro corrects this issue

closed 2024-12-01 by RobertLorincz