LorinczElectronics/Orbiter-Toolboards #16: Orbitool O2 probe current

christianbundy, 2024-11-12:
The documentation indicates that the Orbitool O2 supports the BTT MicroProbe v2, but when I plug it in the probe only turns on the LED. When using other power sources, it turns on the LED and runs a self-test that actuates the servo three times.

What’s the max current for the probe 5V power? My multimeter seems to show ~0.4A, but the probe data sheet shows a max current of 3A. Is this a faulty Orbitool O2, or is the Orbitool incompatible with the MicroProbe v2? Or maybe something else?

smbenson0712, 2024-11-12:
After reviewing the first issue on S03 board and wiring the probe this way - Pin1 - Black, Pin2 - Red, Pin3 - Yellow, Pin4 - Black, and Pin5 - White (Like the pinout diagram with servo and probe swapped) and changing the servo pin to PA15 and Probe Pin to PB3 as suggested, I can get the probe to deploy on power up, but not do the full self test.  It will not retract the probe with the stow macro.  The end stop works as it should, open when deployed and triggered when stowed.  If I wire like Section 3.8, I get no probe movement on power up or via macro, leading me to believe that the PA15 is actually the servo pin.
This is very confusing as the silkscreen on the board shows pin 5 as gnd like section 3.8, The pinout shows ground as pin 4.  Can you please advise on correct wiring and pin configuration for the O2 toolboard.

christianbundy, 2024-11-13:
Thanks! For what it’s worth, my problem occurs even when the only pins connected are +5V and GND, and this is specifically the Orbitool O2. I appreciate the heads up on the similar problem, but for other issues (especially unrelated to these 2 pins or the Orbitool O2) it might be better to create a new issue so that it can be thoroughly documented and easy for others to find!

smbenson0712, 2024-11-13:
Noted.  I am also using the 02 board and having the same problem with power up self test.   My concern was also the wiring discrepancies in the docs  that may also be causing an issue.  Maybe we both have bad boards, as I have tested with 3 new probes, I would say it's not the probes.

christianbundy, 2024-11-13:
On Discord I saw someone say that they had the same problem with the MicroProbe with the Octopus board, which makes me wonder whether this is an issue with the BTT Microprobe requiring lots of current on startup rather than specifically an issue with the Orbitool. It isn't clear to me whether this is as simple as "the buck converter doesn't supply enough current" or whether there's some safety feature that's preventing the MicroProbe's high current draw. (Or, maybe it's something else entirely?)

Happy to test if you have any ideas, I have a bunch of capacitors laying around and thought maybe that would help maintain 5V as the current increases (if that's even the issue), but I'm not great with electronics.

christianbundy, 2024-11-13:
I've opened an issue in the MicroProbe issue describing the issue as well, but it isn't clear to me where the problem is: https://github.com/bigtreetech/MicroProbe/issues/38

christianbundy, 2024-11-14:
I measured with a multimeter today, and when I hook my 5V power supply up to the probe it pulls 2A to deploy the servo. When using the Orbitool, I see a current draw of 0.15 mA, which makes me think it's an issue with the Orbitool rather than the probe.

smbenson0712, 2024-11-19:
I can confirm that the probe it not getting enough power.  It will self test when the probe is in the horizontal or probe up position. When the probe is facing the bed it will not self test completely.  It will deploy, but can't get enough power to retract.  According to the BTT documentation, the probe only requires 350mA to operate , what is the max 5V output from your board on the probe servo pin?

norston, 2024-11-19:
I am having exactly the same issue as you smbenson0712. My microprobe will work just fine in a horizontal or upside down position. I discovered this by accident, BTW. When I was swapping the probe to my spare probe I didn't bother to connect it to the print head, so I was just holding it in my hand, and viola, it worked! My printer is a normal type printer in the sense that it requires the probe pointing downward at the print bed. This problem seems to indicate that the probe isn't getting enough current to lift the probe upwards in it's normal position. I measured my voltage and the circuit is getting 5v. I've tried a couple different Microprobes v2.0 and it's exactly the same issue. These same probes work just fine on my EBB42 in any position. I should also add that the probe is correctly triggering the end stop when the probe is actually able to deploy or retract.

Look, I'm not trying to hijack this issue but since smbenson0712 went down this path I felt compelled to respond. I'll now open my own issue in case this one gets closed.  Thank you.

christianbundy, 2024-11-21:
> According to the BTT documentation, the probe only requires 350mA to operate , what is the max 5V output from your board on the probe servo pin?

My memory is that plugging it in would start between 300 mA and 400 mA and then drop to around 200 mA, but I tried to test today and accidentally fried my board. I’m not sure how. I tried to power a small light bulb but kept getting 0A on my multimeter, and then the board popped and the magic smoke came out.

I’m not sure what happened, but West3D confirmed in Discord that they’d had a few other bug reports from the Orbitool not supporting the MicroProbe, so its unlikely that I’ll be actively working on this problem any longer. I’m out $45 and a too many hours of debugging, but hopefully my flailing around GitHub/Discord/etc at least helped make this problem more apparent.

Good luck y’all!

norston, 2024-11-21:
Thanks for going as far as you did Christian. Sorry to hear about your board. Does anyone have the actual circuit layout for this board?  I’m curious to see how they are bucking down the voltage to 5v. At this point in time I think we need Dr Lorincz to shed some light.

RobertLorincz, 2024-11-27:
The 5V power supply on the toolboard is designed for a constant 500mA load current. It can supply more for shorter time periods. The limitation is only the thermal limitation of the Buck coil and driver circuit. It can supply 1A from more than 1 second.
From my opinion, the issue is the microprobe itself, even if users have positive experience when connecting it to BTT boards. 
I did purchase a microprobe, and I was not able to make it work reliably. Here are my findings and checks:
- Checked the 5V power supply can deliver enough current for the probe - result OK - measured with scope drop over the 5V power supply when deploying the probe;
- checked the probe output signal - result bad behavior - Orbitool have internal pull=up resistor of 10kOhm - the probe output is not able to correctly pull the signal to low logic level - I get something like 2V as low level logic. I disconnected the 10K resistor and only used the internal micro pull up - behavior improved, but still low level was about 1.5V not zero. This explains why the behavior on some BTT board is better. In general, its a very bad idea to use as pull up only the internal resitance of the micro which is realively high impedance. Also, direct connection without any protection circuit is a bad idea. - I'm sure btt boards will fail ESD and EMI immunity test on this function - I doubt they've have done it...or done it correctly...
- lastly the deploy force of the probe is very weak...many cases fails to deploy fully even when supplied and wired correctly - even when I directly connected it to microcontroller pins like on BTT boards.
- I checked the probe accuracy they claim - is nowhere near I have seen up to 0.2mm differences.
- lastly this was only one sample I have tested, maybe my sample was in the bad performance range I donno - however due to the above issues I mentioned, I do not support or recommend the usage of this probe. Simply because I cannot guarantee proper operation of it. I prefer not to support the probe than to remove all the protection features of the board to improve a litle but not fix it totally.