LorinczElectronics/Orbiter-Toolboards #18: Klipper Menuconfig Settings

grainiac, 2024-11-23:
@RobertLorincz is there a detailed description of the required options in menuconfig to select, in order to compile a fresh firmware for the STM32F042? It's very easy to end up with a ROM that won't fit on the MCU'S flash memory. Maybe I was just blind but I did't find. Thank you!

grainiac, 2024-11-23:
The error I am trying to fix is the following (it seems the MCUs need to be on the same version):

MCU Protocol error

This is frequently caused by running an older version of the
firmware on the MCU(s). Fix by recompiling and flashing the
firmware.

Your Klipper version is: v0.12.0-377-g9bd0d4757
MCU(s) which should be updated:
**orbitoolO2: Current version v0.12.0-155-g4cfa266e**
Up-to-date MCU(s):
**mcu: Current version v0.12.0-377-g9bd0d4757**

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.

mcu 'orbitoolO2': Unable to extract params from: config_lis2dw

grainiac, 2024-11-23:
I made some progress and have the Orbiter2 with the Toolboard now working with the firmware that it shipped with: v0.12.0-155-g4cfa266e

I commented the accelerometer related blocks, after that it worked:

###########################################################################################
###################### Onboard accelerometer and resonance tester config ##################
###########################################################################################

#[lis2dw]
#cs_pin: orbitoolO2:PA4
#spi_bus: spi1
#axes_map: y, z, x

#[resonance_tester]
#accel_chip: lis2dw
#probe_points: 140,120,30  # XYZ head position for testing
#min_freq: 10
#max_freq: 200

RobertLorincz, 2024-11-27:
check the user manual; there is an explanation of how to compole klipper. Some unused features need to be deactivated to reduce the generated code size.

closed 2025-01-16 by RobertLorincz