LorinczElectronics/Orbiter-Toolboards #19: How to run a 4028 fan with PWM input on the OS2?

briggs2025, 2024-12-04:
I would like to run a PWM input (three wire) 4028 part cooling fan. I have Tool Fan and Part Fan ports available. We could use either PB1 or PA10 for the PWM output (to the fan). I just need to know how to make one of these ports always on, in order to supply power to the fan. Anybody have success or experience with what I'm trying to do?

RobertLorincz, 2024-12-05:
Could you tell me which partfan type you plan? Usually 3 pin fans have GND, POWER and tacho...in this case, you can leave the tacho signal out or connect it to a spare input if you have, example an input from the BL_Touch sensor.

smbenson0712, 2024-12-10:
Great question briggs2025. I too want to use a 4028. Just to be clear, these can come in three wire in two flavors. One with a tach, or one with a PWM input. Or, you can get them four wire with all of the above. Looking forward to your response Mr Lorincz.

smbenson0712, 2024-12-10:
Need an always on power for these fans, convert Tool Fan or Part fan like you said. Then, make one of these ports output PWM signal to fan. Right?

smbenson0712, 2024-12-10:
Figured it out - Using the Tool Fan port for fan power positive and negative(PB1) and using Part Fan negative pin(PA10) for PWM of fan.  You need to compile the klipper firmware for the toolboard to include Pin PB1 as on at startup in the GPIO Pin section of make menuconfig.  Then use pin PA10 for your part cooling fan pin in klipper config file.

Traveler61, 2024-12-10:
Does the 4028 run full speed at power up and then once booted stop? I have the high speed 4 wire version.

briggs2025, 2024-12-10:
Worked great, thank you smbenson. Mine works great now. Runs full blast until firmware kicks in.

smbenson0712, 2024-12-10:
Traveler61 - the fan will run full speed until the firmware boots up and if you have the PWM pin configured like normal [fan], it will go to zero because that would be it's normal state as a part cooling fan.  I have modified my M106/M107 gcode and scaled the fan output, so that the slicer fan speeds are more normal - the fan only operates between 38%-100% in PWM mode, so the fan settings in the slicer are different.  If you have the 4 wire model with the tach, you could try and hook the 24V, Tach, Gnd to Part fan and PWM to Tool Fan.  Please note : I have not tried that as my fan doesn't have a tach wire.