LorinczElectronics/Orbiter-Toolboards #21: Updating Klipper firmware error

Cagle3545, 2024-12-12:
I am trying to update klipper on my Orbitool O2. I have klipper updated on all other usb devices and on my Pi and Leviathan. I am on klipper version v0.12.0-396-gb7233d11. This is the error message I am getting when I run dfu-util --list :dfu-util: Cannot open DFU device 0483:df11 found on devnum 12 (LIBUSB_ERROR_ACCESS). I can see the board is in DFU mode.  Any help is appriciated.

Cagle3545, 2024-12-12:
I tried to flash the board and this is the message I get in return 

Voron@pi:~/klipper $ make flash FLASH_DEVICE=0483:df11
  Linking out/klipper.elf
/usr/lib/gcc/arm-none-eabi/12.2.1/../../../arm-none-eabi/bin/ld: out/klipper.elf section `.text' will not fit in region `rom'
/usr/lib/gcc/arm-none-eabi/12.2.1/../../../arm-none-eabi/bin/ld: region `rom' overflowed by 385 bytes
collect2: error: ld returned 1 exit status
make: *** [Makefile:73: out/klipper.elf] Error 1

Cagle3545, 2024-12-13:
[image]

[image]

The guide shows support external sensor devices, can the guide be updated to reflect what is needed to be selected.

RobertLorincz, 2024-12-14:
There are changes in the Klipper config optional features section. I will update the documentation soon. Until then, just activate only the features I suggest in the current documentation and deactivate all the rest.

closed 2025-01-16 by RobertLorincz