LorinczElectronics/Orbiter-Toolboards #23: O2 not controlling extruder motor.

MisterMicc, 2025-01-15:
[tmc2209 extruder]
uart_pin: S03:PB5
interpolate: true
run_current: 0.85
hold_current: 0.1
sense_resistor: 0.110
stealthchop_threshold: 0
driver_TBL: 0
driver_HEND: 6
driver_HSTRT: 7
driver_TOFF: 4

new toolboard not spinning my motor.

please chime in if you have any ideas.

RobertLorincz, 2025-01-16:
Your configuration seems to be wrong; check the config file recommendation here:
https://github.com/RobertLorincz/Orbiter-Toolboards/blob/ff7b98a028c4b539d2e272a329198ad768093f6f/Orbiter%20O2%20Toolboard/Klipper%20Config/OrbitoolO2.cfg

MisterMicc, 2025-01-17:
> Your configuration seems to be wrong; check the config file recommendation here: https://github.com/RobertLorincz/Orbiter-Toolboards/blob/ff7b98a028c4b539d2e272a329198ad768093f6f/Orbiter%20O2%20Toolboard/Klipper%20Config/OrbitoolO2.cfg

Thank you, all sorted and printing.

Is it possible to use a bbt Eddy with this toolboard?
I cannot find anything in the documentation

gilbertorconde, 2025-03-17:
> Thank you, all sorted and printing.
>
>
> Is it possible to use a bbt Eddy with this toolboard?
> I cannot find anything in the documentation

I don't think so. Eddy uses CAN or USB, so the tool board would need to be working with CAN (which it isn't) or have a USB expansion hub of some sort on the orbitool (again, it doesn't have one). You can do it by running the USB wires from your RPI (or whatever you are using) to the toolhead alongside to the orbitool USB wire.
Looks to me, the best approach would be to just run the data lines and grab the power (5v, gnd) from the probe pins on the orbitool

RobertLorincz, 2025-03-19:
> Thank you, all sorted and printing.
>
> Is it possible to use a bbt Eddy with this toolboard? I cannot find anything in the documentation

Unfortunately, the current toolboard version does not support such sensors. You need to wire it sepparatelly to your mainboard.