LorinczElectronics/Orbiter-Toolboards #26: Request document with klipper firmware settings and caveats.

coldelectrons, 2025-02-10:
To compile Klipper firmware, certain things need to known:
- mcu type (stm32f042)
- clock reference (???)
- pins for communication interface (???)
- application start offset (???)

Shane-Gadsby, 2025-02-12:
It took me a hot minute to find them too, but it was actually in the manual: https://www.orbiterprojects.com/download/SO3/Orbitool%20SO3%20User%20Manual.pdf page:10

Karlauer, 2025-03-09:
The problem is that the documentation for the O2 and SO3 on github is outdated. 

Document versions on github:
- SO3: 1.13
- O2: 1.12

A lot of vendors direct to the github for detailed documentation. After searching a lot I found newer document version for the O2 (Document version 1.2) as well. These include klipper configuration, but the screenshots on page 11 show the config for the SO3 board. Maybe there is a newer version I haven't found.

I found a shop that offers the O2 manual: https://cdn.shopify.com/s/files/1/0239/9287/files/Orbitool2_v2.72_Manual.pdf?v=1716243417

Pleaes update the documentation on github. Many sites point here for a setup guide.

RobertLorincz, 2025-03-11:
The documentation on Github is the latest. I cannot control what vendors are publishing on their website.

closed 2025-11-17 by RobertLorincz