LorinczElectronics/Orbiter-Toolboards #27: SO2 Communication Issues

winningfaith81, 2025-02-21:
Greetings! I'm having a heck of a time trying to troubleshoot a problem I keep having with my Orbitool SO2 board when I'm losing communication to the board. For what it's worth, I have a humidity of about 30% in my house right now with it being winter and my heat is running.

### **The Error:** 
> Klipper reports: SHUTDOWN
> 
> Lost communication with MCU 'orbitoolO2'
> Once the underlying issue is corrected, use the
> "FIRMWARE_RESTART" command to reset the firmware, reload the
> config, and restart the host software.
> Printer is shutdown

### **What I'm Doing:** 
- Example 1: I'm working on my printer to calibrate my print settings. I go to wipe off excess filament from the hot nozzle with my metal scraper or metal tweezers post print. The printer shuts down and throws the error. 
- Example 2: I need to measure the standoffs I'm using between the Orbiter v2.0 and the Orbitool board. I've got my digital calipers out and I'm measuring the length, touching one side to another. The printer shuts down and throws the error. 
- Example 3: I'm installing the aluminum camera mount to the outside of my aluminum enclosure. The printer shuts down and throws the error. 
- Example 4: I've taken off the Beacon probe. I was trying to get the error to pop. I reached down to home XY and just before or AS I was touching the LCD, the printer shuts down and throws the error. 

### **What I DON'T Experience:** 
I have left the printer on for hours and hours, not using it. I have never witnessed the printer randomly giving the error. It has always happened while I was touching it in some manner. 

### **Troubleshooting:**

1. Checked the ground - I am grounded in all of my electronics into the aluminum frame and to the earth connection that runs into the power from my house. I think I might have two points grounded into the frame, so perhaps this is an issue - creating a cyclical route?? I'm not an electrician. I did verify continuity from the earth ground prong on the IEC inlet with various points in my build like a screw that is put directly into the end center tap of an extrusion, my headed bed ground point, and both of the points where I have grounds screwed on to the frame. All points have continuity. 
2. I have a Beacon probe. I thought perhaps its operation was causing electrical disturbance and me touching the hotend with something metal was causing a short. I disconnected the beacon and messed around with the printer again. Still get a communication loss. 

### **Hardware Setup:** 
- Zero G CoreXY printer. 
- Frame - Aluminum powder coated frames from DLLPDF.com
- Hotend - Phaetus Dragonfly BMO
- Extruder - Orbiter v2.0 with the v2.5 Upgrade kit
- Filament Sensor - Orbiter Filament Sensor v2 w/ tangle detect 
- Toolboard: Orbitool SO2 with stock, unmodified comm cable connected to supplied USB adapter board, connected with stock USB cable to RPi 5 for communication and directly to PSU for power. (For what it's worth, this board does not yet have the 3D printed mount yet. It is screwed directly into frame extrusion. 
- Probe - Beacon Rev H Low profile, connected with unmodified comm cable connected directly to RPi5

### **My Printer File** 
[printer-config.txt](https://github.com/user-attachments/files/18913664/printer-config.txt)

### **Logs**
Klippy Log that shows the error around 11:58am CST 02/21/2025
[klippy.log-02.21.2025-12.07pmCST.txt](https://github.com/user-attachments/files/18913634/klippy.log-02.21.2025-12.07pmCST.txt)

winningfaith81, 2025-02-21:
I wonder if perhaps my issue is similar to this issue with [the fake / bad parts in the USB adapter board](https://github.com/RobertLorincz/Orbiter-Toolboards/issues/12#issuecomment-2509734355) .... I purchased mine from KB3D on 12/17/2024.

cyablo, 2025-03-01:
Hi there, I have the exact same problem on my o2. I was already going crazy and changed the whole Raspberry + power supply and tried different kernels before finding this issue ticket.

RobertLorincz, 2025-03-01:
If the issue happens when touching the printer, it's probably some grounding issue. Have you connected printer GND to the frame and to the earth connection? Do you have maybe exposed wires onto which you can discharge? In some printers with metal parts in the carriages accumulate static charge, and that can discharge at some point, causing all kinds of unexplicable issues.
The second very important topic is to have a stable 5V supply voltage, since the 5V supply, which comes from the RPI USB connection, supplies the board protection electronics. If that 5V supply voltage is not stable or is actually lower than 5V will cause troubles. The latest adapter board design avoids this by having its own stable 5V supply voltage, so it is not influenced by external noise.

cyablo, 2025-03-01:
The issues happened randomly, sometimes after just a few layers, sometimes after hours, sometimes not at all. I never touched the printer. It also happens when printing unattended over night. But lately, it happens almost every print and on the first few layers.

I tried a few power supplies and monitored them with a Fnirsi FNB48 USB tester. The one I use now is stable at ~5,1V, so that also shouldn't be the problem.

Is the problem supply voltage related? Because I dind't change anything but the issue went way worse yesterday and I can't finish a single print since then. Some bad parts were mentioned.

cyablo, 2025-03-01:
Maybe worth mentioning: It's not unusual that the Pi freezes right after klipper tries connecting to the toolboard after booting. I though it was the Pi and changed that too but the problem was still present.

winningfaith81, 2025-03-02:
I've not noticed anything with Pi freezes. 

Robert - I have the USB board wired directly into my power supply. Should I move it to one of the HE ports on my main board? I'm running a BTT Octopus MAX EZ board. My PSU is the stock one from my Ender 5 Pro, it's a clone of Meanwell, a LRS-350-24. 

I have my ground running from the IEC power plug to a wago where it splits to another wago on either side of the electronics bay and grounds into frame. The far side of the printer is also grounded to the frame. The only wires with bare ends are the ones that go into the wagos.

RobertLorincz, 2025-03-02:
Your connections are OK. What I meant is the 5V supplied from the PI. via the USB cable... If that 5V is not stable or lower than 5V can cause issues.

cyablo, 2025-03-02:
It doesn't exactly crash. You can still see the Shell over HDMI and the blinking cursor, but the USB stack seems to crash because keyboard and ethernet stop working at this point, so I can't really tell what is going on when it happenes. Happens on RPI 2 and 3B+ but always boots fine if I pull the toolboard cable before booting the pi.

RobertLorincz, 2025-03-02:
It seems to me your issue is due to the RPI. I have also experienced issues when I used RPI 3A+, sometimes it was starting up with a communication timeout error; it was recovering just fine after commanding a firmware restart. Sems this is due to slow RPI, since this issue never happened to me with RPI4.

cyablo, 2025-03-02:
When the Pi freezes, it's when klipper tries to connect to the orbitool. There is nearly 0 load on the Pi at that time and a Pi 3 has plenty of power running Klipper. Even the Pi 2 I had before is good enough. The freezing also NEVER happens during print when the Pi has some real load. Only on first connection when the Pi boots.

Besides that, my main problem is this:
[image]

RobertLorincz, 2025-03-05:
I know it has quiet enough power to run klipper. I just state what we have observed. And this startup disconnection is more likely with RPI3 and below. I had the same experience; after switching to CM4 and RPI4, the issue was gone. Seems to be related to some kind of task management. It seems during startup some communication is delayed, causing the timeout, but honestly, this is just a theory we were thinking of and never had fount a real smoking gun proof, just obervation. In my opinion, the protocol is too sensitive, especially at startup. I would retry to connect before throwing an error, which would work because every time a manual restart is issued, the problem is solved.

cyablo, 2025-03-05:
Maybe that's the case, I've no RPi 4 >= at hand, so I can't test. maybe I'll try some x86 SFF PC as Klipper host this weekend, if I've got some time, just to see if that resolves the startup connection issue.

But that's a problem I can work around like you said. My main problem really is loosing the connection to the o2 during printing without interaction, like in my screenshot. My reseller I bought the toolboard from said he hasn't heard of an issue like this. What's the best way to contact LDO to get an confirmation that it's a known problem?

cyablo, 2025-03-12:
I shall get a new adapter board from LDO on friday. Will report if that fixes my problem.

dmitry-voronoff, 2025-03-20:
Hi, I have the exact same problem—completely random disconnections from the Orbitool. The GND connection is fine (tested), and the power supply is also fine (I even bought a powered USB hub). I've flashed the latest Klipper firmware onto the Orbitool, BTT Pi2 (not RPi), and Octopus Pro v1.1, so they all have identical Klipper versions. BTT Pi 2 is powered directly from power supply 24V (not usb). However, I’m still experiencing disconnections.

Please help. I can't print!

[klippy.log](https://github.com/user-attachments/files/19373673/klippy.3.log)
[moonraker.log](https://github.com/user-attachments/files/19373808/moonraker.2.log)

winningfaith81, 2025-03-20:
Something I found was that the port where the cable connects to the board might not be connected well enough. Mine came right off recently. I used my high temper solder gun and attached it back to the plate. Testing is still going on to see if the problem persists. [image]

winningfaith81, 2025-03-22:
Additionally, you want to make sure you have some sort of cable strain relief in place, like the board mount that is provided by the creator. I feel that with all the movement, the main cable does not stay seated well at all. There is no catch mechanism between the port and the cable connector like Molex cables have. I think this paired with general movement of the toolhead (and a poorly soldered port) is going to give intermittent connection problems while printing, for sure.

dmitry-voronoff, 2025-03-22:
[image]

[image]

Here is my setup. I think it is completely fine.

cyablo, 2025-03-22:
Just to let you know, I got a new USB adapter board from LDO and I'm sorry to say that it does not work at all. The green LED lights up when connecting USB to the Raspberry, and the O2 Toolboard also powers up (the probe lights up, and the filament sensor initializes) when 24V is applied. However, there is no USB connection with this board—lsusb only shows the SKR E3 mainboard.