LorinczElectronics/Orbiter-Toolboards #28: RGB Leds not working when heater fan is running

The-JurB, 2025-02-26:
I have a voron dragonburner with a Orbitor V2.0, the Orbitor 2 Smart Sensort and the Orbitool O2 (yes i'm using everthing, very good products by the way!)

De dragonburner als has 2 nozzle leds and 1 logo led. The leds are connected to the Orbitool O2. In klipper i am using https://github.com/julianschill/klipper-led_effect as recommended by the dragonburner.

Everything works fine when the printer is idle. I can change colors of the leds, control them individually etc.
It works both fine with the klipper-led_effect macro's and manual setting of the OrbiLed in Fluidd.

However, when the heater fan is running, only the first led of the rgb-chain can be changed. The other 2 leds are not responding. When the heater fan is off, all the leds can be changed.
 This behaviour is the same voor using klipper-led_effect macro's and manual changing the OrbiLed color. Only the first led can be changed. In the klippy log, the following error shown when you update the leds: "Neopixel update did not succeed"

Led configuration in the OrbitoolO2.cfg:

[neopixel OrbiLED]
pin: orbitoolO2:PA1
chain_count: 3 #setup specific
color_order: GRB  #setup specific
initial_RED: 0.0
initial_GREEN: 0.0
initial_BLUE: 0.1

RobertLorincz, 2025-03-01:
Well, this is a strange issue. I personally had not this kind of issue, but sometimes I also have trouble updating colors. Looking into the details, I found out that the timings of signals generated by klipper are really at the edge. Sometimes, depending on the LED you are using, the pulses are even out of spek. We have to open a ticket to klipper developers to fix this...I had one friend who changed the default timings to correct ones, and the issue disappeared. Best would be to have the bit time configurable, as even though all of the leds are WS2812 different variants have slightly different timing requirements.

The-JurB, 2025-03-12:
Thanks for your reaction. Can you open a ticket? You know exactly what to put in their.

dkowis, 2025-04-12:
I have also run into this, with almost the exact setup.

 I have a stealthburner with a rainbowbarf LEDs (so there's 8 LEDs just in that logo, because bling!) and two nozzle LEDs, and the LEDs just wedge when that fan comes on.

I also see the klippy log "Neopixel update did not succeed." so yeah I think I'm in the same predicament.

https://forge.light.kow.is/3d-printing/v350_config/src/branch/main/conf/sb_leds.cfg#L59-L76

```
[neopixel sb_leds]
pin: orbitoolO2:PA1
chain_count: 10

color_order: GRB,GRB,GRB,GRB,GRB,GRB,GRB,GRB,GRBW,GRBW

initial_RED: 1.0
initial_GREEN: 0.45
initial_BLUE: 0.0
initial_WHITE: 0.0
```

gilbertorconde, 2025-05-15:
My leds start failing as described when i replaced my toolhead fan from a normal 2 pins 24v one to a 3 pins (tacho). I could always go back, but i think the 3th pin in my fan is more important than the leds. But would be cool to understand and fix this. I've an O2 as well. Now im using a 3010 axial fan dc24v from a creality k1

RobertLorincz, 2025-05-17:
I would say this is some kind of weird Klipper issue; the RGB LEDs have no feedback...klipper does not know what they are set...so if it gives an error, probably it was not able to configure it.

Lemaro2211, 2025-07-09:
I only have a USB camera and my two toolboards connected, and I'm running Klipper on a Dell Wyse 5070, so the USB bandwidth shouldn't be capped yet. I tried connecting another LED strip directly to my SKR 1.3, and the LEDs work better there — although still not 100%. The first LED is always on, and I can only set certain colors like blue or white. Basically, everything else doesn't work at all.

dkowis, 2025-07-09:
I can change LEDs usually, until the hotend fan fires up. As soon as that happens, I can no longer set the LEDs. I have to set up the stuff _before_ firing up the hotend. I can do animated things without any hotend fan running, and that seems to work reliably, but there's definitely some kind of timing issue happening when the toolhead is "busy." 

I don't pretend to know enough about the guts of how all this works, I can only describe the symptoms.

Lemaro2211, 2025-07-09:
I am able to change the color of the strip connected to the SKR, even when the toolhead is heated and the fans are running.

dkowis, 2025-07-09:
Yeah, I have no symptoms on my other LED strips attached directly to my Octopus. They always worked, even when I had a toolhead directly attached to that board.

Lemaro2211, 2025-07-09:
@dkowis Lucky you! XD I don’t know what it is, but none of the LED strips I’ve tested with Klipper have worked properly. However, all of them work perfectly with WLED, so the LEDs themselves are fine.

RobertLorincz, 2025-07-17:
> Any updates on this? On my O2 Board, the LEDs are white all the time, and I can’t even turn them off. I love the board so far, but this is really annoying. I have tried changing the BIT_MAX_TIME bit with no luck. For some reason, the LEDs seem to have some sort of addressing — since my strip is 22 LEDs long, if I set it to 22, all of them light up, but if I set it to 6, only 6 light up.

I do not have any updates so far. It is strange you have that issue; in my case, I tested, and I can update RGB LED's even when the hotend is running. One thing pops up in your text: you are using 22 LED;s connected to the RGB output? I'm certain the toolboard onboard power supply is not able to drive so many LED's The supply current is limited to 0.5A max as written in the user manual. A typical RGB LED consumes 60 60mA when commanded max white - with 22 LEDs, this would mean 1.32A. I'm not aware of any bit time configuration possibility for Klipper; as far as I understand, this is hardcoded.... It would be nice if the bit time would be configurable so we can adapt the signals to different LEDs as I learned not all WS2812 versions have the same bit timing.

Lemaro2211, 2025-07-17:
I am using COB LEDs. They have a much higher LED density but are not as bright. The strip I bought has 200 LEDs per meter and draws 12 W/m, meaning a single LED only draws 5 mA, and the 22 should draw around 0.11 A. I thought that would be fine.

Lemaro2211, 2025-07-17:
Pushing the limit how? Power-wise, or is it a problem sending out the data packets to so many LEDs? Because I have a simple ESP32 controlling around 800 LEDs with no problem (WLED, just for reference).

RobertLorincz, 2025-07-25:
Indeed there is an issue in Klipper's RGB signal timing. The timing is hardcoded, and the default is not optimum for all RGB versions, and in addition, we do not get the same timing on different micros. In the RGB connector power supply line there is a current limiter. If the current is too high, it will reduce the supply voltage or enter into protection mode. To have a correct operation of the RGB signals, we need a stable power supply. This does not explain why the issue only appears when the hotend fan is ON...These circuits are completely separate and not even close to each other on the PCB board. Could it be you have some power supply wiring issues on the RGB LEDs? Maybe GND not connected correctly, or you have some GND loops?