LorinczElectronics/Orbiter-Toolboards #29: SO3 Toolboard Not Detected by Raspberry Pi

jjwb22101, 2025-03-24:
Brand new SO3, connected to 24V power and USB, is not being detected by my printer's raspberry pi. I run lsusb (or ls /dev/serial/by-id/) and the main board MCU shows up just fine (BTT Octopus), but there are no results indicating that the SO3 toolboard is detected at all. It's definitely receiving power, since I get green LEDs on both the toolboard itself and the adapter board, and the fan on the SO3 spins up to max speed.

I've tried connecting it to all four of the USB ports on the Pi (the main MCU shows up fine regardless of port), swapping around USB cables (again, no difference), and even just plugging it into a USB port on my laptop. No luck on any of those. Any suggestions?

RobertLorincz, 2025-03-24:
Have you tried re-programming the board, using the RESET and BOOT buttons?

jjwb22101, 2025-03-24:
Yup. No changes regardless of what combination of buttons I push. Doesn't show up regardless

jjwb22101, 2025-04-03:
Any suggestions on other things I could try? Or is the board likely DOA?

JoeTheTacoLord, 2025-04-19:
I am having this same exact issue, is there any update on this? I ordered a second tool board and still had the same issue, so I doubt its just DOA. Is there supposed to be any LED indication on the board when it enters DFU mode?

RobertLorincz, 2025-04-22:
There is no LED showing up when in DFU mode. Usually if you supply the board with the 24V and USB connected, the DFU mode shall show up. If it already has Klipper loaded, then it should respond with a serial number. And DFU mode can be activated by RESET and BOOT buttons. Press down both buttons, release RESET and after about 2-3 seconds, release BOOT. and the unit should be in DFU mode.

closed 2025-11-17 by RobertLorincz