LorinczElectronics/Orbiter-Toolboards #30: All of a sudden, fans not working anymore on Orbitool O2

technik-gegg, 2025-03-24:
I'm currently working on my own tool changer upgrade, which will eventually be equipped with 4 independent tool heads. Hence, I bought 2 Orbitool O2 tool board kits. 
I installed both of them and applied the example Klipper configuration file. I was able to do the first calibration steps, i.e. PID tuning on both successfully and even got some filament to extrude for the very first crude test. 
After a couple of days while I was working on the printer, I went to calibrate the E-Steps for both extruders. While on the 1st one everything went as expected, on the 2nd one I got skipping steps because the nozzle got clogged. Evaluating the issue, I figured that this was because the hotend cooling fan wasn't spinning (anymore). So I checked my configuration first but everything was ok. After that I went and measured the voltages on the tool board. What I found is, that all of a sudden, the hotend fan spits out a constant 12V (although on a 24V system) after powering the machine. 
On the other two (Tool Fan and Part Fan) nothing happens. These two I've configured as generic fans in order to be able to spin them up manually but no voltages whatsoever. Everything else on this tool board is working as it's supposed to (i.e. heater, thermistor, stepper motor and even the Neopixel LED).
Just to be sure, I flashed the tool board once again with the latest Klipper driver but to no avail. On a short visual check I didn't spot anything concerning (so no burned ICs or similar). I even checked the PCB traces for the Tool fan and everything seems just fine.
Is this a (maybe rare but) known issue? 
Is there something I can do or check or is this tool board now dead and has to be replaced?

RobertLorincz, 2025-03-24:
Hi. It can be issues with the board; however, I cannot judge this based on your desciption. I assume you have different serial number set for the two boards, and pin configuration and MCU name is updated accordingly.

technik-gegg, 2025-03-25:
Robert,
thank you for your quick reply. I found the culprit and as always, it was sitting in front of the monitor :o)
After you mentioned the configuration, I checked it again. 
Yes, I did rename all the sections, i.e. mcu became orbitoolO2_0 and orbitoolO2_1 / tool_fan_0 and tool_fan_1 but even though I looked over it probably a couple of hundred times, I somehow managed to overlook the [heater hotend_fan], which I didn't rename.
Just after starring at the config files for quite some time and comparing them side by side it eventually hit me... go figure!

I'm glad it's nothing wrong with the tool boards and I'm now able to continue my journey.
At least my post might be helpful for others facing similar issues in the future.

TG

closed 2025-03-25 by technik-gegg