LorinczElectronics/Orbiter-Toolboards #32: Klipper "step_pin" error message

Pete2554, 2025-04-16:
I believe I have an issue with 3 of my 4  So3 circuit boards. In each case Klipper comes up with the error message " Option 'step_pin' is not valid in section extruder x". I have reinstalled the the configuration files for each extruder and have redone the DFU process. Figuring it might have something to do with the USB cables, I swapped out, in turn, each USB cable assembly including the pi to USB cable, power board along with the USB cable to the extruder circuit board from the working extruder. With each cable assembly swap the other extruders were commented out in printer.cfg. Everything points to an issue or defect with the extruder circuit boards. The boards are essentially brand-new and have only recently been installed.

Below is the relevant section from the Klippy log:

=======================
temperature_probe btt_eddy: No drift calibration configured, disabling temperature drift compensation
temperature_probe btt_eddy: registered drift compensation with probe [probe_eddy_current btt_eddy]
Config error
Traceback (most recent call last):
  File "/home/makertech/klipper/klippy/klippy.py", line 130, in _connect
    self._read_config()
  File "/home/makertech/klipper/klippy/klippy.py", line 127, in _read_config
    pconfig.check_unused_options(config)
  File "/home/makertech/klipper/klippy/configfile.py", line 497, in check_unused_options
    self.validate.check_unused(config.fileconfig)
  File "/home/makertech/klipper/klippy/configfile.py", line 446, in check_unused
    raise error("Option '%s' is not valid in section '%s'"
configparser.Error: Option 'step_pin' is not valid in section 'extruder3'
webhooks client 548070833408: New connection
webhooks client 548070833408: Client info {'program': 'Moonraker', 'version': 'v0.9.3-72-g7cdcca3'}
===============================

Thanks in advance and hoping you can help.

RobertLorincz, 2025-04-22:
Could you share your configuration? Out of the description, I cannot figure out what your issue is. It seems you have something misconfigured.

Pete2554, 2025-04-22:
Hi Robert,

Thank you for your reply.

As per your request please find attached one of the extruder configuration files that's producing the "step_pin" error message. I've also attached the complete Klippy log file just in case.

Best regards,
Pete

________________________________
From: LorinczElectronics
Sent: Tuesday, April 22, 2025 7:53 AM
To: RobertLorincz/Orbiter-Toolboards
Cc: Pete2554; Author
Subject: Re: [RobertLorincz/Orbiter-Toolboards] Klipper "step_pin" error message (Issue #32)

[https://avatars.githubusercontent.com/u/155807093?s=20&v=4]RobertLorincz left a comment (RobertLorincz/Orbiter-Toolboards#32)<https://github.com/RobertLorincz/Orbiter-Toolboards/issues/32#issuecomment-2821410103>

Could you share your configuration? Out of the description, I cannot figure out what your issue is. It seems you have something misconfigured.

RobertLorincz, 2025-04-23:
I cannot find the attachments, the klippy log is not interesting

Pete2554, 2025-04-23:
Hi Robert,

I forgot that email firewalls don't like configuration files. Please see the zipped attachment.

I assume the extruder configuration is the only one you need.

Best regards,
Pete

________________________________
From: LorinczElectronics
Sent: Wednesday, April 23, 2025 1:11 PM
To: RobertLorincz/Orbiter-Toolboards
Cc: Pete2554; Author
Subject: Re: [RobertLorincz/Orbiter-Toolboards] Klipper "step_pin" error message (Issue #32)

[https://avatars.githubusercontent.com/u/155807093?s=20&v=4]RobertLorincz left a comment (RobertLorincz/Orbiter-Toolboards#32)<https://github.com/RobertLorincz/Orbiter-Toolboards/issues/32#issuecomment-2825284580>

I cannot find the attachments; the klippy log is not interesting

closed 2025-05-12 by Pete2554