LorinczElectronics/Orbiter-Toolboards #36: orbitool02 with BTT Eddy coil

gilbertorconde, 2025-05-10:
I'm wondering if it's possible to use the i2c version of the BTT Eddy. Now I have a cartographer but would be nice to just have one USB cable on my umbilical.

I noticed the eddy-ng project added touch probing capabilities to these probes.
So if I could connect to i2c version directly to the probe pins of the orbitool I could have all of this with just one cable.

I still have a lot of investigation to do on the BTT Eddy side (make sure the i2c version can use touch, how accurate it is, how it is affected by thermal drift if at all, etc) but this would be a good start.

cyablo, 2025-05-15:
That would be cool, but as far as I know, the used pins for the Probe PA15 and PB3, don't offer I2C on the STM32F042 Microcontroller.

gilbertorconde, 2025-05-15:
but [here](https://github.com/RobertLorincz/Orbiter-Toolboards/tree/main/Orbiter%20O2S%20Toolboard) it says i2c for bed level sensing and scanning. Am i missunderstanding anything?

closed 2025-05-15 by gilbertorconde

cyablo, 2025-05-16:
Oh, I didn't even notice there will be a o2s Version with I2C and USB aux. Great! I was looking into the NiteHawk for my 2nd printer because I also wanted to use an eddy probe. Maybe I'm gonna wait for the o2s now.

gilbertorconde, 2025-06-16:
> [@gilbertorconde](https://github.com/gilbertorconde) can you tell, which pin you used for sda and which for scl? Thanks. Unfortunally there are still no schematics for the O2 available.

Sure, im using these:
```
scl_pin: toolhead:PB3 # Servo signal port (Remember, dont use alias)
sda_pin: toolhead:PA15  # Limit signal port
```

But what do you mean by "unfortunally there are still no schematics for the O2 available"? dont you have the schematics in this repo? If you're trying to say that there's no documentation on how to use the probe in O2 i dont think it will ever be. But you can get there by yourself pretty easly. If you follow the documentation on the probe they have an example on how to use the pins from the bltouch, then you can check the documentation on the O2 board and match them.
For the O2 you actually dont need it cause i just gave them to you, but if you have any other board you wish to test is basially the same.

gilbertorconde, 2025-06-16:
oh, ok. i get your point, sure. Well, at least we know we can use the pandapi sensor one. Btw, i've been printing  reliably with the probe. Having no issues at all.
Important notes:
Pin alias won't work, apparently they seem working, no immediate failures, but when start using the probe it start failing.
The bed mesh can't be as detailed as with cartographer usb. I'm using 20x20 probe count. Seems to be the maximum. (with cartographer i was able to go as far as 50x50).

qOtXqi, 2025-06-22:
@gilbertorconde @cyablo 
I can confirm the BTT Eddy Coil is working fine with the Orbitool O2 (non s).

`[probe_eddy_current btt_eddy]
i2c_mcu: orbitoolO2
i2c_software_scl_pin: orbitoolO2:PB3
i2c_software_sda_pin: orbitoolO2:PA15`

even a 50x50 rapid_scan with 200 mm/s is working.

gilbertorconde, 2025-06-22:
That's quite amazing. I really thought that wouldn't work. Now I'm considering switching it. Many tks for the config. Is there anything particularly different that needs to be done when flashing the board? (Any new entry on the optional features or so?)

qOtXqi, 2025-06-22:
@gilbertorconde 
Your welcome. When flashing the OrbitoolO2 you need to enable the optional features
```
[*] Support communicating with external chips via I2C bus
[*]     Support software based I2C "bit-banging
[*] Support ldc1612 eddy current sensor
```
and whatever else you need.

cyablo, 2025-07-03:
Really cool that it's working. I also ordered an Eddy Coil to try myself (only got the USB Version lying around, waiting for the O2S to be released or a NH36 for me to buy). I didn't know software I2C in Klipper is already fully implemented and stable.

Did you experience any problems so far?

qOtXqi, 2025-07-03:
@cyablo No, none. Everything is working fine. Even the load on the Orbitool STM32F042x6 is comfortably far from 100% while scanning. It has a load of 65% and an awake time of 86%.

gilbertorconde, 2025-07-03:
> Really cool that it's working. I also ordered an Eddy Coil to try myself (only got the USB Version lying around, waiting for the O2S to be released or a NH36 for me to buy). I didn't know software I2C in Klipper is already fully implemented and stable.
> 
> Did you experience any problems so far?

i've been using as well since the last message i posted here and its working flawless since. The best part is its working even better than the pandapi. More reliable and less resource hungry. With pandapi, whenever i do a full bedmesh scan my cameras stop working during the process so you can see how heavy it is. With the BTT eddy coil it just works.
After you set up all the things (the z-offset that works for you) it just works.
I find myself just sending jobs to my voron without even going to baby check the first layer.

cyablo, 2025-11-08:
> > [@gilbertorconde](https://github.com/gilbertorconde) Your welcome. When flashing the OrbitoolO2 you need to enable the optional features
> > ```
> > [*] Support communicating with external chips via I2C bus
> > [*]     Support software based I2C "bit-banging"
> > [*] Support ldc1612 eddy current sensor
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > and whatever else you need.
> 
> Im using now the eddy coil as well, with eddy-ng. Works just perfect, and on top of it the TAP is way lighter with this. The BD sensor really pushes a bit to hard for my taste. I just had a problem when building the firmware with the eddy-ng installed but with a minor change in the `sensor_ldc1612_ng.c` i was able to put it to work. ~Im pretty sure it is because i have used a cartographer and never uninstalled that, the eddy-ng probably search in you installation and, if it sees that code uses a different configuration.~ Again, tks for the compilation info
> 
> **EDIT:** the change in the `sensor_ldc1612_ng.c` is needed because the developers in the eddy-ng project made a wrong assumption on how to identify the type of sensor. More here: [vvuk/eddy-ng#66 (comment)](https://github.com/vvuk/eddy-ng/issues/66#issuecomment-3001065791)

Sorry to resurrect this issue like this, but: I also wanted to try eddy-ng now. How did you get klipper + eddy-ng to compile with the needed menuconfig settings? The firmware it produces is a bit too big:  region `rom' overflowed by 355 bytes.

gilbertorconde, 2025-11-08:
> Sorry to resurrect this issue like this, but: I also wanted to try eddy-ng now. How did you get klipper + eddy-ng to compile with the needed menuconfig settings? The firmware it produces is a bit [...]

Its a good question, that happened to me as well. I just went over all the options and disabled everything I wasn't using especifically on this board. I don't remember exactly what options I disabled but go over them and try to see by the name if this board uses it. The menuconfig is genetic for any board that uses this chip so there's somethings that we don't need. If I have the time I'll try to go over it again and post it here.
In the meantime if you find it yourself please post it as well for future reference

cyablo, 2025-11-09:
Yep, looks like Neopixel isn't needed when not using the RGB header. Firmware fits now and everything still seems to work, did a few test prints with ender-ng now working :)

gilbertorconde, 2025-11-10:
For future information I have a toolhead working with the RGB leds and the Smart sensor. There's other things that can be disabled.
This is how my toolhead looks like:
[image]

[image]

[image]

Edit: thanks to this photos I noticed that my hotend isnt straight 🤣 it's amazing how well it prints like this

gilbertorconde, 2025-11-13:
ok, i just reflashed my orbitool O2 with this:
```
[*] Support micro-controller based ADC (analog to digital)
[*] Support communicating with external chips via SPI bus
[ ]     Support software based SPI "bit-banging"
[*] Support communicating with external chips via I2C bus
[*]     Support software based I2C "bit-banging"
[*] Support GPIO based button reading
[*] Support Trinamic stepper motor driver UART communication
[*] Support 'neopixel' type LED control
[*] Support measuring fan tachometer GPIO pins
    *** LCD chips ***
[ ] Support ST7920 LCD display
[ ] Support HD44780 LCD display
    *** Accelerometer chips ***
[ ] Support adxl accelerometers
[*] Support lis2dw and lis3dh 3-axis accelerometers
[ ] Support MPU accelerometers
[ ] Support ICM20948 accelerometer
    *** External ADC type chips ***
[ ] Support thermocouple MAX sensors
[ ] Support HX711 and HX717 ADC chips
[ ] Support ADS 1220 ADC chip
    *** Other external sensor chips ***
[*] Support ldc1612 eddy current sensor
[ ] Support angle sensors
```
Everything seems to be working. I do have neopixel leds and the smart filament sensor